jrkerns/pylinac #325: Leeds TOR incorrectly Phantom Angle When Phantom ~0deg

randlet, 2020-11-20:
**Describe the bug**
When a Leeds phantom is imaged at roughly 0 deg Pylinac incorrectly determines the phantom angle to be ~350deg.

**To Reproduce**
Run Leeds Tor analysis (with invert=True) on attached image.

**Expected behavior**

Angle should be detected as 0deg.

**Screenshots**

The issue is due to the way the circular profile looks when the phantom is at ~0deg.  In this case the peak from the lead square gets split between the left and right half of the profile like this:

[image]

A second problem is that the LeedsTOR._is_counterclockwise() function doesn't always work correctly if the [roll here](https://github.com/jrkerns/pylinac/blob/master/pylinac/planar_imaging.py#L772-L774) produces detected peaks like this:

[image]

A more robust method is to roll by the FWXM idx.

PR Incoming!

randlet, 2020-11-20:
Example file.  (txt added to file name so GitHub lets me upload it).
[kv-tor18fg-dynrange.dcm.txt](https://github.com/jrkerns/pylinac/files/5574396/kv-tor18fg-dynrange.dcm.txt)

jrkerns, 2020-11-20:
Thanks. Looks like this phantom also must be force-inverted due to the closed blades. I guess I never added an open-blade restriction in the docs. I should add that.

randlet, 2020-11-20:
Yes, needs to be inverted (as mentioned in the OP!).

jrkerns, 2020-11-20:
> Yes, needs to be inverted (as mentioned in the OP!).

You know I don't read issues 🤨

closed 2020-11-20 by jrkerns