jrkerns/pylinac #328: Thanks :)

SimonBiggs, 2020-11-20:
Hi @jrkerns,

Just wanted to say thank you for this commit:

https://github.com/jrkerns/pylinac/commit/25ba0ff72ccfbd5517a44df48d9f841fbc15e2e6

It fixed my issue over here :)
https://github.com/pymedphys/pymedphys/pull/1217#issue-524969526

jrkerns, 2020-11-21:
Awesome. Needed this today. Thanks. 😃

closed 2020-11-21 by jrkerns

SimonBiggs, 2020-11-21:
> Needed this today

I'm going to make sure to say thanks more often then, cause there's a lot
of things I have to be thankful for from you :).

jrkerns, 2020-11-21:
@SimonBiggs you may find the following helpful if you're still doing WL work: https://github.com/jrkerns/pylinac/blob/85ec7d8d69c88d32188e1b5eda8ff374493554c9/image_generator/generate_dcm.py

Usage is similar to keras where you add layers:
```
from pathlib import Path
import pylinac
from image_generator.generate_dcm import *

# generate fake WL images
for i in range(10):
    path = Path("generated_images", f"WL-{i}.dcm")
    im = AS1000Image()
    im.add_layer(FieldLayer((30, 30)))
    im.add_layer(BBLayer(5))
    im.add_layer(GaussianLayer(sigma_mm=1))
    im.add_layer(RandomNoiseLayer(sigma=2000))
    im.generate_dicom(path, gantry_angle=i*36)

# analyze them using the WL module, just to see
wl = pylinac.WinstonLutz(Path("generated_images"))
print(wl.results())
wl.plot_summary()
```

SimonBiggs, 2020-11-21:
Ahh yes! That is awesome. Have you seen the hypothesis library (https://hypothesis.readthedocs.io/en/latest/) that would be awesome combined with this :).

I shall dive into this this week. Thanks @jrkerns :).

SimonBiggs, 2020-11-21:
I'd be keen to be able to show you the fruit of the WLutz labour once it's done. Still probably a few more weeks until it is "plug and play". So I'll wait till then before doing show and tell.

jrkerns, 2020-11-22:
I've seen hypothesis but never used it. Seems like a good fit for a more generic library like my hidden baby [argue](https://gitlab.com/jrkerns/argue). Thanks

SimonBiggs, 2021-01-08:
If you'd like to check it out, I've managed to get the beginnings of my WLutz Arc code working within the online app:

https://share.streamlit.io/pymedphys/pymedphys/streamlit-app-staging/app.py

If you go to the above URL and then click on "Winston-Lutz Arc" it should work. ... it's not the best demo data, I was trying to see how much I could break it... I'm going to swap the demo data out.

[image]

There's certainly quite a bit more work needed on it. In particular creating an Excel file report for downloading. But, you can see the bones of what it will become 🙂

jrkerns, 2021-01-08:
I'm always impressed with your velocity in making new things. Looks pretty awesome!

SimonBiggs, 2021-01-08:
Thanks @jrkerns 🙂

SimonBiggs, 2021-02-09:
Thought you might also be interested in this little update:

https://share.streamlit.io/pymedphys/pymedphys/streamlit-app-staging/app.py

If you go to that app above and click on "DICOM WLutz":

[image]

You should be able to drag and drop DICOM files onto the UI and then have it use PyLinac to calculate the field centre and bb centre, and then use PyMedPhys to provide a feedback report for each field.