jrkerns/pylinac #334: Catphan module : save_analyzed_subimage issue

jlley, 2021-02-02:
Hello,

I've updated to pylinac v2.4.0.
I analyze capthan 504 images for clinical purpose.
I have the following issue with the new pylinac version:

 ```
 File "C:\Users\p0093469\.conda\envs\qatrack0.3.update\lib\site-packages\pylinac\sanstitre0.py", line 19, in <module>
    mycbct.publish_pdf('mycatphan.pdf', open_file=True)  # open the PDF after saving as well.

  File "C:\Users\p0093469\.conda\envs\qatrack0.3.update\lib\site-packages\pylinac\ct.py", line 1150, in publish_pdf
    self._publish_pdf(filename, metadata, notes, analysis_title,

  File "C:\Users\p0093469\.conda\envs\qatrack0.3.update\lib\site-packages\pylinac\ct.py", line 1170, in _publish_pdf
    self.save_analyzed_subimage(data,img)

  File "C:\Users\p0093469\.conda\envs\qatrack0.3.update\lib\site-packages\argue\argumentative.py", line 50, in wrapper
    raise error_type(message.format(

OptionError: Argument 'sp' passed to 'save_analyzed_subimage' was not one of the allowed options: ('hu', 'un', 'lc', 'mtf', 'lin', 'prof')
```

It seems to come from the function publish_pdf. I use the function as described in the documentation:  
`mycbct.publish_pdf('mycatphan.pdf', open_file=True) `

Any idea to resolved this issue?

Thank you,

Jean-Luc

jrkerns, 2021-02-09:
Hi Jean-Luc, 
    Do you have v2.4.0.1? I think that was a silly bug I found after releasing 2.4.0. Just update to the latest pylinac from pypi: https://pypi.org/project/pylinac/

Let me know if you still have trouble after upgrading.

jlley, 2021-02-11:
Hi James,

Yes it resolved the issue.
Thank you for your help.

Jean-Luc

closed 2021-02-12 by jrkerns