jrkerns/pylinac #338: Picket Fence Module - HDMLC Parameter Missing

bvangenderen, 2021-02-23:
Running the PicketFence analyze() function in v2.4.0.3 raises a TypeError.

Followed the method outlined in the documentation to analyze a picket fence image:
`
from pylinac import PicketFence
pf = PicketFence('PATH TO DICOM FILE')
pf.analyze(0.1, hdmlc=True)
`
which failed with the following:
`TypeError: analyze() got an unexpected keyword argument 'hdmlc' `

I downgraded to Pylinac v2.4.0.1 and the issue disappeared and the analysis completed as expected.

When I looked at the analyzed image when using the v2.4.0.3 analysis without the hdmlc parameter, it didn't look like the analysis was matching the HDMLC so I don't think the function is working as intended, but I thought I should check.

slizzered, 2021-03-04:
I believe the API for that function has changed. Instead of supplying the `hdmlc` parameter to `analyze()`, you can set the type of MLC directy when creating the PicketFence object:
``` Python
from pylinac import PicketFence
pf = PicketFence('PATH TO DICOM FILE', mlc='HD')
pf.analyze(0.1)
```

This allows you to precisely define the MLC arrangement, instead of just a simple ON/OFF switch for the HD-MLC.
In the file `picketfence.py`, you can find a dictionary with all pre-defined MLC examples:

``` Python
MLCs = {
    'Millennium': MLCArrangement([(10, 10), (40, 5), (10, 10)]),
    'HD': MLCArrangement([(10, 5), (40, 2.5), (10, 5)]),
    'Halcyon Distal': MLCArrangement([(60, 5)], offset=2.5),
    'Halcyon Proximal': MLCArrangement([(60, 5)]),
}
```

bvangenderen, 2021-03-04:
Ah, thank you! I hadn't seen the documentation for the new API yet but this looks very flexible!

jrkerns, 2021-03-06:
Unfortunately, this happened because of my sloppy git flow practices. This new functionality was supposed to be for 2.5. It will definitely be more flexible. Bit of a change in MLC position calculation, but overall this will be better for flexibility. I better get 2.5 wrapped up then...

bvangenderen, 2021-03-08:
No worries! Looking forward to the new release. :)

closed 2021-03-08 by bvangenderen