jrkerns/pylinac #347: read binary Halcyon file (to CSV)

angioedema, 2021-03-16:
Hi,
We would like to read the Halcyon bin file and convert it in CSV. To do this, we use the TrajectoryLog class and convert to csv like this : 
`tlog_path = r"bin.bin"
tlog = TrajectoryLog(tlog_path)
tlog.to_csv()
`
However, we have a problem in the CSV. For instance, for the Gantry Actual in units of degrees field, I have a lot of values. These values are write in the CSV file, in one cells for several and the cells below  for another part ect. But, the last value of the first cells is truncated ... and the first value of the second cells, below, is the continuation of the value, less a number. 
Following an example : 
First cell (last values) : ........ 193.50018310546875,193.59561157226562,193.6859130859375,**1**
Second cell (first values) : **3.7708282470703**,193.87152099609375,193.9556121826172,194.054443359375 .................
Here, you could see the truc and the 9 is missing
Expected : ........ 193.50018310546875,193.59561157226562,193.6859130859375,**193.7708282470703**,193.87152099609375,193.9556121826172,194.054443359375 .................

Is it a bug or we have missing a step ?
BR

jrkerns, 2021-03-16:
Unfortunately, Varian does not publish the tlog specifications for Halcyon like they do for TrueBeam. This means it's a trial and error experience to get compatibility with Halcyon. I have not tried parsing a Halcyon file, but if you could upload a bin file for me with known values (doesn't have to be exact, just rough descriptions and context of the delivery) to give a sanity check to the values it would go a long way to improving the log parsing module. You can upload files here: https://www.dropbox.com/request/YKRu4AmuPsXu55uQq761. Thanks!

angioedema, 2021-03-17:
Thank you. I have put the file. But I do not understand which kind of informations you want, do you please be more specific ?
Thanks again !

jrkerns, 2021-03-17:
I need to know some of the axis information to know if my parsing module is correct. E.g. the gantry start and stop angle, collimator angle, couch position, a rough shape of the fluence, how long it took to deliver, etc. When parsing a .bin file you have to know the format ahead of time and if you don't then it's almost a guessing game to figure it out.

angioedema, 2021-03-19:
Hi,
I work with the Physicien that would like to read the bin. He give me the associated anonymised Dicom, I put it on the Dropbox and he give me this to : arc 1 = 70s et arc 2 = 64s --> 134s total. Is it okay, or you need something else ?
Thanks ;)