jrkerns/pylinac #354: Catphan Modules without background_roi_settings result in nan background_median & background_std values

randlet, 2021-03-31:
**Describe the bug**

For Catphan modules without background_roi_settings (CTP486) the `background_median/background_std` get set to nan in [_setup_rois](https://github.com/jrkerns/pylinac/blob/master/pylinac/ct.py#L251) because np.mean([]) == np.std([]) == nan.  This throws RuntimeWarnings and / or FloatingPointError's.  This nan gets propagated through to cnr calculations for the module rois

**To Reproduce**

```python

from pylinac import CatPhan604

cbct = CatPhan604.from_zip("CatPhan604.zip")
cbct.analyze()
print([roi.cnr for roi in cbct.ctp486.rois.values()])

# /lib/python3.6/site-packages/numpy/core/fromnumeric.py:3373: RuntimeWarning: Mean of empty slice.
# /lib/python3.6/site-packages/numpy/core/_methods.py:170: RuntimeWarning: invalid value encountered in double_scalars
# /lib/python3.6/site-packages/numpy/core/_methods.py:234: RuntimeWarning: Degrees of freedom <= 0 for slice
# /lib/python3.6/site-packages/numpy/core/_methods.py:195: RuntimeWarning: invalid value encountered in true_divide
# /lib/python3.6/site-packages/numpy/core/_methods.py:226: RuntimeWarning: invalid value encountered in double_scalars

# [nan, nan, nan, nan, nan]

```

randlet, 2021-03-31:
Also, somewhat related, should

```python
            background_median = np.mean([roi.pixel_value for roi in self.background_rois.values()])
```
be
```python
            background_mean = np.mean([roi.pixel_value for roi in self.background_rois.values()])
```
or
```python
            background_median = np.median([roi.pixel_value for roi in self.background_rois.values()])
```
?

jrkerns, 2021-03-31:
Quantitatively, mean or median doesn't make a big difference in this context, but for consistency, let's go with mean (middle).

closed 2021-04-05 by jrkerns