jrkerns/pylinac #366: WLT ROI manipulation (trick/Tipp)

Kiragroh, 2021-04-13:
Hi,

can you please help me with a specific problem.

The left outside Panel of our Imager is defect (white column in test Images). Not a big problem for PortalDosimetry because it is not in center and can be mostly countered with calibration but WLT failed because pylinac things that the BB has to be in the column.

Is there a easy way to define/force a ROI in the center of a image? Or would you try to implemented code to ignore over exposured regions for automatic ROI finding?

thank you

jrkerns, 2021-04-13:
I will be adding extra checks to the WL module to verify the BB is near the center of the image in the next version. For the time being, I would crop the images in an external program if possible.

Kiragroh, 2021-04-14:
Thanks that you will work on that. Maybe this WLT-Images (with defect Panel region) will help with testing. Kind regards

https://ukede-my.sharepoint.com/:f:/g/personal/m_grohmann_uke_de/EjO65um6WDtKgH_YYXzBYnYBiPebHA9ZZTvza-6lQqHfGg?e=ZgH260

commit 5e6d7566 (2021-06-12): Closes #333 Closes #366
commit 6a1cce94 (2021-06-12): Merge pull request #396 from jrkerns/bugfix/366_wl_bb_near_center

Closes #333 Closes #366
closed 2021-06-12 by jrkerns