jrkerns/pylinac #368: Catphan CTP404 slice thickness

jmbm94, 2021-04-14:
Hi, 
I've a problem with the parameter, slice thickness, in Catphan600's analysis. The correct value should be double of the value that I obtain. 
I suposse that the problem is in the function:
      ' def meas_slice_thickness(self):
        return np.mean(sorted(roi.wire_fwhm*self.mm_per_pixel*0.42 for roi in self.thickness_rois.values()))/(1+2*self.pad)'

Thanks!

jrkerns, 2021-04-14:
Hi jmbm94, 
     Can you provide a little more information? Have you tried in another QA program? What are the actual values? If you know the correct value then it sounds like you don't need any measurements; you can simply double the value pylinac provides.

jmbm94, 2021-04-15:
Yes, I do. I tried it with AutoQA Lite and this parameter is introduced in CT by me, so I know the actual value. 
It's a bit strange because, using the same protocol and phantom but different acquisitions, the result sometimes is evaluated correctly but other times gives a wrong value. The different acquisition were performed without changing the phantom position.
Sorry about the previous post, when I said 'The correct value should be double of the value that I obtain', it was wrong. I have performed more acquisitions and the measurement of the slice thickness is not always the half value of the nominal value.