jrkerns/pylinac #370: MultiProfile doesn't detect all peaks for some cases

randlet, 2021-04-15:
While trying to debug issues related to #323 I tried using a MultiProfile to find the peaks of a profile but only the right hand side peaks are found.

**To Reproduce**

```
import numpy as np
import matplotlib.pyplot as plt
from pylinac.core.profile import MultiProfile
prof_data = np.array([0.01491,0.00923,0.01014,0.00758,0.00601,0.00335,0.00499,0.00049,0.00440,0.00402,0.00000,0.00237,0.00369,0.00449,0.00254,0.00323,0.00430,0.00932,0.00675,0.00659,0.00565,0.00553,0.00197,0.00548,0.00467,0.00485,0.00669,0.00377,0.00486,0.00390,0.00821,0.00256,0.00302,0.00594,0.00576,0.00749,0.00459,0.00563,0.00529,0.00729,0.00598,0.00481,0.00701,0.00632,0.00648,0.00511,0.00915,0.00826,0.00668,0.00508,0.00872,0.00468,0.00520,0.00875,0.01141,0.01115,0.00681,0.00760,0.00584,0.00878,0.00574,0.01002,0.01081,0.00754,0.01161,0.01091,0.00690,0.01044,0.00790,0.01113,0.01180,0.00843,0.01019,0.00738,0.01068,0.01135,0.01197,0.00908,0.01025,0.01529,0.01267,0.01253,0.01415,0.01333,0.01219,0.01264,0.01482,0.01424,0.01098,0.01629,0.01180,0.01586,0.01424,0.01613,0.01562,0.01615,0.01711,0.01476,0.01607,0.01530,0.01485,0.01435,0.01750,0.01474,0.01505,0.01811,0.01979,0.01779,0.01567,0.02049,0.01814,0.01950,0.01905,0.01757,0.02093,0.01992,0.01942,0.01786,0.02251,0.02197,0.02363,0.02368,0.02345,0.02310,0.02316,0.02121,0.02285,0.02418,0.01962,0.02261,0.02775,0.02388,0.02274,0.02535,0.02336,0.02517,0.02545,0.02637,0.02847,0.02220,0.02618,0.02763,0.02770,0.02729,0.02751,0.03110,0.03061,0.02728,0.02963,0.03049,0.03677,0.02982,0.03031,0.03385,0.03327,0.03113,0.03840,0.03169,0.02959,0.03135,0.03165,0.03234,0.03059,0.03162,0.03333,0.03549,0.03593,0.03452,0.03673,0.03975,0.04010,0.03778,0.03860,0.03541,0.04154,0.04526,0.04346,0.04022,0.04357,0.04286,0.03886,0.04200,0.04562,0.04688,0.04333,0.04568,0.04611,0.04830,0.04860,0.04941,0.05112,0.04950,0.04688,0.05169,0.05247,0.05599,0.05057,0.05495,0.05455,0.05585,0.05492,0.05489,0.05418,0.05623,0.05948,0.05596,0.06067,0.05725,0.05656,0.06137,0.06230,0.06001,0.06789,0.06750,0.06873,0.06561,0.06878,0.06679,0.06939,0.07094,0.06956,0.06865,0.07507,0.07695,0.07758,0.07947,0.07635,0.07881,0.07817,0.08168,0.08066,0.08750,0.08951,0.08956,0.09403,0.09756,0.09836,0.10364,0.11207,0.11606,0.13136,0.14360,0.17319,0.21420,0.27017,0.34291,0.43074,0.51912,0.59940,0.66895,0.71914,0.75979,0.79461,0.82302,0.84414,0.86432,0.87917,0.88823,0.90490,0.90748,0.90929,0.91120,0.91984,0.92563,0.92033,0.92047,0.92811,0.93138,0.93057,0.92998,0.93393,0.93219,0.93583,0.93588,0.93852,0.93740,0.93998,0.94129,0.94125,0.94296,0.94218,0.94401,0.94555,0.94693,0.94797,0.94827,0.94535,0.94694,0.94635,0.94998,0.95167,0.94514,0.95030,0.95079,0.94888,0.94767,0.95399,0.95439,0.95494,0.95493,0.95140,0.95095,0.95331,0.95148,0.95608,0.95442,0.95307,0.94187,0.93457,0.92416,0.90101,0.87616,0.82514,0.77242,0.70212,0.64445,0.59238,0.54527,0.51944,0.50459,0.50768,0.52760,0.55669,0.61121,0.67446,0.74421,0.80153,0.84270,0.88067,0.90494,0.92629,0.93850,0.95481,0.95641,0.96155,0.96595,0.96374,0.97067,0.96878,0.96962,0.96970,0.96919,0.97645,0.96998,0.96934,0.97360,0.97454,0.97462,0.97598,0.97358,0.97298,0.97817,0.97477,0.97594,0.97211,0.97224,0.97579,0.97560,0.97539,0.98010,0.97653,0.97373,0.97551,0.97673,0.97852,0.98019,0.97952,0.97597,0.97340,0.97704,0.97843,0.97967,0.97844,0.97946,0.98172,0.97744,0.98169,0.98023,0.97937,0.98098,0.98240,0.97941,0.98134,0.98151,0.98459,0.97673,0.96922,0.95919,0.94367,0.91764,0.87786,0.81635,0.75865,0.69718,0.64366,0.60288,0.58055,0.57436,0.59134,0.61957,0.66381,0.72676,0.79101,0.84136,0.88317,0.91702,0.93982,0.96234,0.97088,0.98207,0.98571,0.99144,0.99101,0.99433,0.99289,1.00169,0.99790,0.99723,0.99491,0.99719,0.99440,0.99738,0.99417,0.99356,0.99863,0.99137,0.99877,0.99802,0.99390,0.99984,0.99792,0.99966,0.99734,0.99934,0.99935,0.99857,0.99926,0.99776,1.00068,0.99345,0.99644,0.99816,1.00274,0.99921,0.99627,0.99942,0.99695,1.00130,0.99848,0.99747,0.99926,1.00116,0.99957,1.00356,0.99916,0.99520,1.00139,1.00164,0.99853,0.99484,0.99746,0.99472,0.98657,0.98807,0.97782,0.95381,0.93318,0.88648,0.83269,0.76805,0.70941,0.65783,0.62600,0.60773,0.61239,0.63159,0.66487,0.72092,0.77709,0.83096,0.88081,0.91773,0.93769,0.95865,0.97268,0.98237,0.99506,0.99709,1.00309,0.99923,1.00123,1.00246,1.00098,1.00194,1.00302,0.99844,1.00329,1.00166,1.00201,1.00114,1.00755,1.00013,1.00268,0.99887,1.00206,1.00373,1.00128,1.00114,1.00123,0.99973,1.00594,1.00026,1.01341,1.00985,1.00857,1.01068,1.01152,1.00921,1.00880,1.00978,1.01043,1.00511,1.00809,1.00773,1.00831,1.00763,1.00805,1.00975,1.00611,1.00282,1.00835,1.00893,1.00084,1.00377,1.00202,1.00621,1.00854,1.00528,1.00188,0.99189,0.98923,0.96973,0.94672,0.91004,0.85751,0.80287,0.73356,0.68287,0.64153,0.61424,0.60974,0.61586,0.65428,0.70480,0.75759,0.81830,0.86616,0.90744,0.93563,0.95557,0.97409,0.98687,0.99115,1.00165,1.00515,1.00315,1.00742,1.00508,1.00932,1.00324,1.00791,1.00428,1.00448,1.00638,1.00557,1.00523,1.00559,1.00607,1.00742,1.00528,1.00634,1.00493,1.01065,1.00708,1.00685,1.00454,1.00896,1.01115,1.00487,0.99927,1.00558,1.00426,1.00820,1.00317,1.00500,1.00237,1.00569,1.00533,1.00674,1.00023,1.00318,1.00486,1.00284,1.00393,1.00558,1.00007,1.00476,1.00380,1.00148,1.00189,1.00038,1.00091,0.99960,0.99975,0.99899,0.99079,0.98891,0.98058,0.95696,0.92267,0.88183,0.82630,0.75437,0.69568,0.64693,0.60724,0.59809,0.60178,0.62125,0.66117,0.71724,0.77794,0.83461,0.87944,0.90947,0.93846,0.95763,0.97429,0.98374,0.98292,0.99368,0.99117,0.99319,0.99051,0.99585,0.99567,0.99931,0.99074,0.99205,0.99256,0.99551,0.99751,0.99015,0.99047,0.98997,0.98935,0.99144,0.99298,0.98964,0.99391,0.99220,0.99243,0.98929,0.99169,0.99697,0.99119,0.99398,0.98875,0.99218,0.99095,0.98985,0.98900,0.98755,0.99036,0.98539,0.98607,0.98652,0.98738,0.98277,0.99100,0.98810,0.98885,0.98223,0.98853,0.98844,0.98245,0.98236,0.98268,0.98215,0.97811,0.98165,0.97812,0.96758,0.96115,0.94658,0.92133,0.87684,0.82911,0.76161,0.69568,0.63606,0.59055,0.57022,0.56308,0.57604,0.61123,0.65385,0.71508,0.77822,0.83008,0.86808,0.90194,0.92691,0.94416,0.94734,0.96020,0.97026,0.96995,0.97358,0.97401,0.97328,0.97233,0.96820,0.97034,0.97140,0.97142,0.96940,0.96588,0.96785,0.96981,0.96820,0.96739,0.96785,0.96465,0.96819,0.96077,0.96107,0.96105,0.96155,0.95988,0.96340,0.96111,0.96259,0.95899,0.95960,0.95786,0.95488,0.95779,0.95463,0.95818,0.95370,0.95177,0.95317,0.94847,0.95019,0.94329,0.94367,0.94232,0.94249,0.93768,0.93714,0.92870,0.92743,0.92122,0.90625,0.89291,0.87103,0.84980,0.81930,0.76410,0.70860,0.64675,0.57540,0.49814,0.42861,0.36547,0.30572,0.26423,0.22150,0.18833,0.16431,0.14868,0.12975,0.12041,0.10923,0.10587,0.09804,0.09711,0.09255,0.09735,0.08823,0.08877,0.09218,0.08114,0.08124,0.08151,0.08016,0.07606,0.07756,0.07840,0.07562,0.07238,0.06727,0.06834,0.07112,0.06955,0.06688,0.07000,0.06800,0.06980,0.06517,0.06225,0.06566,0.06255,0.06737,0.06305,0.06022,0.05983,0.06444,0.05896,0.05899,0.05597,0.06057,0.05492,0.05820,0.05518,0.05467,0.05359,0.05721,0.05156,0.05064,0.05206,0.05180,0.05260,0.05014,0.05066,0.04874,0.04930,0.04881,0.04997,0.04738,0.04235,0.04839,0.04659,0.04135,0.04627,0.04194,0.03951,0.04077,0.04326,0.04289,0.03951,0.04293,0.04337,0.03703,0.04010,0.03399,0.03667,0.03460,0.03819,0.04160,0.03734,0.04170,0.03368,0.03566,0.03377,0.03733,0.03341,0.03449,0.03880,0.03474,0.03106,0.03223,0.03313,0.02927,0.03228,0.03025,0.02951,0.03103,0.03233,0.03127,0.02924,0.02817,0.02670,0.02653,0.02659,0.03196,0.03030,0.02781,0.02697,0.02971,0.02543,0.02447,0.02721,0.02341,0.02532,0.02431,0.02255,0.02281,0.02314,0.02155,0.02196,0.01964,0.02096,0.02448,0.01749,0.02255,0.02082,0.02038,0.01833,0.01832,0.01910,0.01858,0.02100,0.01721,0.01722,0.01709,0.01639,0.01987,0.01804,0.02238,0.01927,0.02047,0.01582,0.01577,0.01640,0.01643,0.01528,0.01800,0.01794,0.01455,0.01587,0.01499,0.01501,0.01569,0.01512,0.01426,0.01484,0.01193,0.01593,0.01077,0.01149,0.01346,0.01856,0.01214,0.00911,0.01110,0.01019,0.01043,0.01106,0.01277,0.01347,0.01320,0.01106,0.01044,0.00684,0.01075,0.00574,0.01156,0.00907,0.00930,0.01415,0.00926,0.00689,0.00660,0.00848,0.00694,0.00755,0.00951,0.00740,0.00872,0.00686,0.00827,0.00556,0.00738,0.00719,0.00529,0.00633,0.00327,0.00711,0.00741,0.00587,0.00960,0.00290,0.00432,0.00961,0.00397,0.00666,0.00431,0.01023,0.00535,0.00621,0.00307,0.00442,0.00310,0.00370,0.00420,0.00635,0.00484,0.00152,0.00449,0.00548,0.00759,0.00371,0.00455,0.00505,0.00350,0.00212,0.00584,0.00634,0.00494,0.00635,0.00218,0.00775,0.00440,0.00791,0.00521,0.01009,0.01477,0.01787,0.03098,0.04860,0.06929,0.09178])

mp = MultiProfile(prof_data)
mp.find_peaks(threshold=0)
mp.find_valleys()
mp.plot()
plt.show()

```

**Screenshots**

[image]

jrkerns, 2021-04-15:
Using v2.5.0 I don't get the same result. What version/commit is this?
[image]

randlet, 2021-04-15:
Lol, I'm an idiot sorry.  Wrong branch.

closed 2021-04-15 by randlet