jrkerns/pylinac #375: CT results_dict Hard to parse value

crcrewso, 2021-04-20:
https://github.com/jrkerns/pylinac/blob/90f261b2754c7a5f7f2f1bcefc822bc339c43109/pylinac/ct.py#L1303-L1305

`myCBCT['CTP404 HU ROI values']` is hard to parse because the returned object is a list of dictionaries each with a single key 

``` python
[{'Air': {'avg value': -1000,
   'cnr': 132.84803425137127,
   'difference': 0,
   'nominal value': -1000,
   'passed': True}},
 {'PMP': {'avg value': -197,
   'cnr': 19.10859745277429,
   'difference': -1,
   'nominal value': -196,
   'passed': True}},
 {'50% Bone': {'avg value': 843,
   'cnr': 54.19025396606719,
   'difference': 118,
   'nominal value': 725,
   'passed': False}},
 {'LDPE': {'avg value': -103,
   'cnr': 13.453974403842945,
   'difference': 1,
   'nominal value': -104,
   'passed': True}},
 {'Poly': {'avg value': -43,
   'cnr': 8.55850922459852,
   'difference': 4,
   'nominal value': -47,
   'passed': True}},
 {'Acrylic': {'avg value': 107,
   'cnr': 5.02771443249051,
   'difference': -8,
   'nominal value': 115,
   'passed': True}},
 {'20% Bone': {'avg value': 243,
   'cnr': 16.105276294623636,
   'difference': 6,
   'nominal value': 237,
   'passed': True}},
 {'Delrin': {'avg value': 341,
   'cnr': 24.376930822432026,
   'difference': -24,
   'nominal value': 365,
   'passed': True}},
 {'Teflon': {'avg value': 998,
   'cnr': 48.62001294814177,
   'difference': -2,
   'nominal value': 1000,
   'passed': True}}]
```

crcrewso, 2021-04-20:
Should the line be 

``` python
{name: {'avg value': val.pixel_value, 'cnr': val.cnr, 
                                         'difference': val.value_diff, 'nominal value': val.nominal_val, 
                                         'passed': val.passed} for name, val in self.ctp404.rois.items()}
```

jrkerns, 2021-06-09:
The results_data will be a hierarchy of dataclasses. I'm about to release a release candidate for v3.0. Accessing data will be like so:

`data = ct.results_data()`
`data.ctp404.hu_rois['Air'].value`

or as a dict:
`data = ct.results_data(as_dict=True)`
`data['ctp404']['hu_rois']['Air']['value']`

Let me know how this structure works for you and reopen if need be. Release candidate coming shortly.

closed 2021-06-09 by jrkerns