jrkerns/pylinac #384: Leeds TO _is_counterclockwise causing unrequired flip

callumgillies, 2021-04-30:
**Describe the bug**
Leeds Test Object image if flipped unnecessarily. Results are then incorrect 

**To Reproduce**
Steps to reproduce the behavior:
Import attached image of Leeds Test Object and plot_analyzed_image
(Unable to upload as dcm so uploaded as jpg. Can supply dcm if required)

**Expected behavior**
I expect the analyze function "_is_counterclockwise" to return false, but it doesn't.
The image is flipped for the analysis which causes the low contrast ROIs to be reversed and the MTF ROIs to be incorrectly placed.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

[image]

**Additional context**
Add any other context about the problem here.

jrkerns, 2021-04-30:
Have you tried forcing inversion? ``.analyze(..., invert=True)``. This might help. https://pylinac.readthedocs.io/en/latest/planar_imaging.html#troubleshooting

callumgillies, 2021-05-04:
I think the problem is the "flip" rather than the "inversion"? - for some reason the "_is_counter_clockwise" function is not not working for any images we are acquiring.
Please find attached the result if I set invert to True, I didn't know if there was a way I could force flip/noflip using the library as is?
[image]

jrkerns, 2021-05-04:
I see some artifacting on the top of the image. If that's true, it would be worth it to crop the image a bit first. Edge artifacts are the # 1 reason for failure of analysis. 

You can crop like so: 
```
leeds = pylinac.LeedsTOR(...)
leeds.image.crop(pixels=5)
leeds.analyze(...)
```

If that doesn't work, then the last resort is a hack to override the method:

```
class MyLeeds(LeedsTOR):

     def _is_counter_clockwise(self):
              return True  # or false, whatever you need

leeds = MyLeeds(...)
leeds.analyze(...)
```

jrkerns, 2021-06-09:
Assuming resolved.

closed 2021-06-09 by jrkerns