jrkerns/pylinac #392: Unable to locate Catphan 604

t-milan, 2021-05-27:
Hello,

The CatPhan module is unable to handle the following CBCT:
[THORAX.zip](https://github.com/jrkerns/pylinac/files/6550290/THORAX.zip)

I've opened it in ImageJ and it seems to me there's nothing wrong with the CT set

jrkerns, 2021-05-27:
The end of the CT appears cut off (red line in figure is where it roughly cuts off). From slice 0-4 it's just blank, then suddenly the phantom is in view. The error is because pylinac is looking for the CT on slice 3, which is blank. The total length of the scan appears fine. Never seen this before. 
[image]

t-milan, 2021-05-28:
Ah, ok. Perhaps a dodgy setup - other CBCTs on the same linac with same settings have processed fine. However it would be ideal if Pylinac still produced results for scan sections 1-3 in this case

closed 2021-06-16 by t-milan