jrkerns/pylinac #402: The 'argue' dependency appears to be depreciated and Pylinac doesnt recognise the module

morganHealy, 2021-08-05:
I having issues importing the Pylinac module becuase the argue module available on https://pypi.org/ doesnt work

jrkerns, 2021-08-05:
Hi Morgan, 
     Are you using an older version of argue? I've not updated that library in quite some time. The most recent code has options in it: https://gitlab.com/jrkerns/argue/-/blob/master/argue/argumentative.py#L5. You can try upgrading with ``pip install argue --upgrade``. You should get v0.3.1 from pypi. Let me know how it goes.

morganHealy, 2021-08-06:
Hi James,

I tried using pip but it's a work computer and I think the firewall blocked the installation. I got the following response in command line:

> WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', timeout('_ssl.c:1106: The handshake operation timed out'))': /simple/argue/

After about 5 retries it terminated saying:

> ERROR: Could not find a version that satisfies the requirement argue (from versions: none)
> ERROR: No matching distribution found for argue

Nevertheless I am able to download repositories from GitHub and Python modules don't require administrator privileges, so I was able to install Argue eventually. After un-zipping I had to copy and paste the python.exe and python39.dll files into the expanded folder, at the same directory as _setup.py_, and then open command line and CD into that directory before running _python setup.py install_ . This was repeated for all depndencies of Pylinac.

For some reason, Argue was more difficult and I also had to copy the entire expanded folder into _C:\Users\morgan.healy\AppData\Local\Programs\Python\Python39\Lib\site-packages\_ and I also had to move _-init-.py_ and _argumentative.py_ into their parent folder (same directory as setup.py). This took a lot of trial and error to figure out.

closed 2021-08-06 by morganHealy