jrkerns/pylinac #417: Starshot: all tested images returns the same result.

effelle, 2022-05-02:
**Describe the bug**
We are trying to implement pylinac in our workflow but sadly we weren't unable to archieve solid results (and I'm pretty sure is our fault!).

**To Reproduce**
Using the following script form docs:
```python
from turtle import clear
from scipy import ndimage

import pylinac
from pylinac.core.image_generator import GaussianFilterLayer, FilteredFieldLayer, AS1200Image, RandomNoiseLayer

star_path = 'C:/Users/effel/Downloads/radioterapia/Images/CR.1.3.51.0.7.2652200017.39611.53065.42883.31209.9902.52590.dcm'
as1200 = AS1200Image()
for _ in range(6):
    as1200.add_layer(FilteredFieldLayer((270, 5), alpha=0.5))
    as1200.image = ndimage.rotate(as1200.image, 30, reshape=False, mode='nearest')
as1200.add_layer(GaussianFilterLayer(sigma_mm=3))
as1200.generate_dicom(file_out_name=star_path)

# analyze it
star = pylinac.Starshot(star_path)  
star.analyze()
print(star.results())
star.plot_analyzed_image()
```

**Expected behavior**
The analyzed image should return different values

**Screenshots**
[image]
[image]
[image]

**Additional context**
I'm quite sure we are missing something since I didn't find open issue for the starshot.
Plese note the script always returns some warnings with opur images:
```
C:\Users\effel\AppData\Local\Programs\Python\Python310\lib\site-packages\pydicom\valuerep.py:290: UserWarning: The value length (36) exceeds the maximum length of 16 allowed for VR SH.
  warnings.warn(msg)
C:\Users\effel\AppData\Local\Programs\Python\Python310\lib\site-packages\pydicom\valuerep.py:290: UserWarning: The value length (18) exceeds the maximum length of 16 allowed for VR SH.
  warnings.warn(msg)
```
I can provide the Dicom files used for the test if needed.  
Thank you in advance for you help!

F.

jrkerns, 2022-05-02:
F, 
   I don't know if you are trying to analyze your own images or perform benchmarking as your scripts show. 

Just to clarify, your scripts are generating artificial images and processing them. I can't tell any difference in code between your screenshots; maybe I missed it. You can use a [gist](https://gist.github.com/) to dump code into. 

If you are trying to process your OWN images then drop the image generator part of the script and just follow the typical use pattern: https://pylinac.readthedocs.io/en/release-3.0/starshot_docs.html#typical-use

If you are trying to produce artificial images then you *shouldn't* expect differences between them. 

In the future, you can post on the [pylinac google group](https://groups.google.com/g/pylinac) for discussions like this.

effelle, 2022-05-02:
Hi James,
the first screenshot is there just to show how the beaming stars are different between the images. The other 3 following screenshot show how the result is the same even if the input image is not. I do apologize if was not clear. 
If I get it well I shuld use this script instead:
```python
from pylinac import Starshot
star_img = 'C:/Users/effel/Downloads/radioterapia/Images/CR.1.3.51.0.7.2652200017.39611.53065.42883.31209.9902.52590.dcm'
mystar = Starshot(star_img)
mystar.analyze(radius=0.5, tolerance=0.8)
print(mystar.results())
mystar.plot_analyzed_image()
```
But the result is always the same:
```
Result: PASS 

The minimum circle that touches all the star lines has a diameter of 0.020 mm. 

The center of the minimum circle is at 639.4, 639.5
```
As the image produced:
[image]

I'm starting to ask myself if I can use a Dicom image as input or if I must use a TIF format.

F.

effelle, 2022-05-02:
So after some trials and errors I found what it seems to work for us:
```python
from pylinac import Starshot
img_path = 'C:/Users/effel/Downloads/radioterapia/Images/COLTESTE02.tif'
mystar = Starshot(img_path, dpi=105, sid=1000)
mystar.analyze()
print(mystar.results())
mystar.plot_analyzed_image()
```
Using tif files, not Dicom.

closed 2022-05-02 by effelle