jrkerns/pylinac #424: Winston Lutz Feature Request

seandomal, 2022-08-25:
For some Facilities, Winston Lutz is performed with the Varian Isocube were imaging actually reveals a dark bb rather than a light one. The current module cannot locate the bb for this circumstance and I would like to request an update to the thresholding algorithm so I can find the bb. This would be a great addition and very useful in the clinic. Thank you!

Specifically I'm talking about the function: _find_bb in the Winston-Lutz module.

jrkerns, 2022-08-26:
Good point. I need to add an `invert=True` parameter to the analyze method, which should address this. Would you mind uploading a set of these images I can test. https://forms.gle/kvbwZ4Q9dfvi8BtT6

seandomal, 2022-08-26:
Just uploaded the files generated for this past monthly QA using the varian isocube, this will be great to be able to address this, thank you!

seandomal, 2022-09-06:
Any updates on this? @jrkerns

jrkerns, 2022-12-28:
Closing d/t being stale/no response

seandomal, 2023-02-09:
@jrkerns I wanted to revisit this issue and provide some additional images. These images don't appear to have any weird artifact but don't seem to be found using the current thresholding in your winston lutz script
[test.zip](https://github.com/jrkerns/pylinac/files/10701642/test.zip)

jrkerns, 2023-02-13:
Got it, thanks for the dataset. I'll investigate.

seandomal, 2023-02-28:
@jrkerns have you had a chance to review those images?

jrkerns, 2023-02-28:
Yes. I tried using a Hough circle detection algorithm as that isn't inversion-specific. However, it didn't pan out as it wasn't precise enough. I think I may have to simply add a parameter to `analyze` to invert the BB detection algorithm.

seandomal, 2023-03-21:
Wonderful, thank you!

jrkerns, 2023-03-21:
Now in v3.9

closed 2023-03-21 by jrkerns