jrkerns/pylinac #430: Dose Linearity from

crcrewso, 2022-12-21:
**Is your feature request related to a problem? Please describe.**
Real world problem - We noticed that the inital dose rate for one of our Linacs changed. This can be seen easily in the SNC Profiler software but not easily quantitatively recorded. 
[image]

**Describe the solution you'd like**
An additional metric added to the SNC Profiler package and the corresponding outputs to produce the relevent aapm metrics

jrkerns, 2023-03-20:
I didn't measure this in the clinic so I'm unfamiliar with the AAPM metrics. Can you point me to the report please? I'm guessing this can only be done w/ .prm files, not .prs files.

crcrewso, 2023-03-21:
[AAPM Practice Guideline for Linacs](https://aapm.onlinelibrary.wiley.com/doi/10.1002/acm2.12080 )
[TG 142](https://aapm.onlinelibrary.wiley.com/doi/full/10.1118/1.3190392)
Table III - Annual Dosimetry - X-ray monitor unit linearity: limits for MU linearity as ±5% (2–4 MU), ±2% (> 5 MU). 

I'm not sure if the information would be in PRS files. I can't find any in our records to inspect.

jrkerns, 2023-03-22:
Hmm. I never thought about doing it that way. We did it with an ion chamber and delivered X MU and compared the normalized result. Is this with a known MU? Or do you integrate the area?

crcrewso, 2023-03-22:
I think that beam was 100 MU. 
I think that the generally accepted way to consider linearity and initial beam quality has changed within the last 10 years, the referenced extension of TG142 dates to 2017. [CPQR guidance from 2016 ](https://www.cpqr.ca/wp-content/uploads/2017/02/MLA-2016-07-03.pdf) also includes considerations for what the document describes as end effect. (AL9 & 10)

jrkerns, 2023-03-23:
I'm not seeing implementation details there? In the US at least, in TG-198 (published 2021) an IC reading is still the default way. 2.D.1.8.3

mchamberland, 2023-03-23:
I'm also curious about using the Profiler to calculate MU linearity, but I don't understand how you would do it or what metric you're calculating from that graph?

@crcrewso Do you have a procedure you can share that you are currently using to calculate it?

crcrewso, 2023-03-27:
We are currently exploring how best impliment this clinically, I haven't seen consensus on how to report this detail; it appears to me to be a recommended tolerance with no suggested implimentation. We are keeping track and manually comparing the first MUs of a 100 MU field to the same field from the commissioning dataset to ensure that the starting peak isn't too extreme but would like to turn this into a quantitiative measure going forward.

crcrewso, 2023-03-27:
For context here is the dose rate for the same field on a different linac, the spike is over the first 10 MU of the field. 

[image]

And an electron field 

[image]