jrkerns/pylinac #433: DeviceFieldAnalysis.results_data() fails with AttributeError for Profiler

StephenTerry, 2023-03-03:
**Describe the bug**
After loading and analyzing a Profiler file using DeviceFieldAnalysis, calling the results_data() method gives an AttributeError because there is no central_roi attribute.

**To Reproduce**
Steps to reproduce the behavior:
1. Load a Profiler .prm file using DeviceFieldAnalysis, then run analyze().
2. Call results_data().
3. Receive error: AttributeError: 'DeviceFieldAnalysis' object has no attribute 'central_roi'

**Expected behavior**
Returning the results_data dictionary.

**Additional context**
Adding a central_roi class to the DeviceFieldAnalysis object with attributes max, mean, std, and min set to None fixes the issue, but I'm not sure if that is the proper way to address this.

jrkerns, 2023-03-20:
Confirmed. Fix should be in the next release due April 20th. Thanks for reporting.

jrkerns, 2024-07-23:
Fixed in 3.10

closed 2024-07-23 by jrkerns