jrkerns/pylinac #440: IC Profiler symmetry value different than Profiler software calculation

StephenTerry, 2023-04-17:
**Describe the bug**
Symmetry calculations for IC Profiler measurements disagree from those calculated by the Profiler software.  This seems to be due to the profiles stored in the horiz_profile and vert_profile variables are not symmetric about the CAX. This results in incorrect points being matched together when the profile is reversed in the symmetry calculation.

**To Reproduce**
See gist at https://gist.github.com/StephenTerry/36c1dbffca05319687c33ab5571a8234
Output shown below.

**Screenshots**
[image]

**Additional context**
Adding an additional point to the right of the CAX by editing the "field values" key in the data dictionary of SingleProfile makes the pylinac symmetry calculation match the Profiler software calculation. I've created a pull request with this edit.

jrkerns, 2023-04-18:
Thanks for the explanation and proposed fix. Can you send me the profiler file so I can reproduce locally? Thanks!

jrkerns, 2023-04-27:
Closed in #439