jrkerns/pylinac #443: Can not import CatPhanBase

arai0101, 2023-05-11:
**Can not import CatPhanBase**
When I try to import CatPhan, there is an error:

Traceback (most recent call last):
  File "linac.py", line 3, in <module>
    from pylinac import CatPhan504
  File "D:\pylinac_train\venv\lib\site-packages\pylinac\__init__.py", line 12, in <module>
    from .acr import ACRCT, ACRMRILarge
  File "D:\pylinac_train\venv\lib\site-packages\pylinac\acr.py", line 21, in <module>
    from .ct import CatPhanBase, CatPhanModule, Slice, ThicknessROI, get_regions
  File "D:\pylinac_train\venv\lib\site-packages\pylinac\ct.py", line 1471, in <module>
    class CatPhanBase:
  File "D:\pylinac_train\venv\lib\site-packages\pylinac\ct.py", line 2048, in CatPhanBase
    def results(self, as_list: bool = False) -> Union[str, list[list[str]]]:
TypeError: 'type' object is not subscriptable

I am using python 3.8, pylinac 3.10.0, with virtualenv. Thank you in advance

jrkerns, 2023-05-11:
Ah, sorry about that. That's a python type thing. The quickest fix is to go to the `ct.py` file and add `from __future__ import annotations` to the very first line of the file and that should fix it.

jrkerns, 2023-05-30:
Fixed in 3.11.

closed 2023-05-30 by jrkerns