jrkerns/pylinac #444: For equate_image function, could the zoom_factor variable be < 1 and resize the biggest image?

LuisOlivaresJ, 2023-05-15:
https://github.com/jrkerns/pylinac/blob/5014284103371a2734650f1f9c9552a48a9c021e/pylinac/core/image.py#L104-L107

If we requiere ```zoom_factor``` to be < 1, and apply ```interpolation.zoom``` to the biggest image **to reduce it**, two things could be gained, optimization and fidelity data:

      1.- Optimization because we are goint to work with less points (for example when calculating gamma).
      
      2.- ​Fidelity data because we are goint to have averaged instead interpolated points.

The idea is relevant when working with TIF files with high resolution (for example 300 DPI versus more standar values as 25.4 DPI). 
 
(This is my first pylinac issue, so I will be grateful for any feedback. I have read pylinac [contributing section](https://pylinac.readthedocs.io/en/latest/contributing.html) and a [guide](https://opensource.guide/how-to-contribute/) for open source contributions, so I hope to be doing the right way).

jrkerns, 2023-05-18:
Hi Luis, 
    Thanks for the suggestion. As it currently stands, the function is meant to zoom image2 to the same level as image1. The documentation isn't very clear on this but image1 is meant to be the reference image. image2 is then scaled to whatever zoom it needs to be to match image1. 
    
   Your points might be valid for your particular use case, but optimizing/shrinking the data feels like it should be the responsibility of the user. They might not want such behavior. As for 2) my experience has been the opposite: downsampling images was a no-no. So I guess this comes down to personal preference. 
   
  To get the same behavior you seem to be wanting, you can simply pass the lowest resolution image first into the function.

LuisOlivaresJ, 2023-05-18:
Hi James, 
Thank you for the feedback. You are right, my suggestion could be relevant just for a particular case, and fortunately the function give us the possibility to do that (just changing the parameter order).

If it is ok, I am going to close the issue.
Thanks again.

closed 2023-05-19 by LuisOlivaresJ