jrkerns/pylinac #446: For equate_images, crop only when it is necessary.

LuisOlivaresJ, 2023-05-16:
I have a bug when I use equate_image for two image that actually have the same physical size but different resolution.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

LuisOlivaresJ, 2023-05-16:
I have added proposition inside commit #445.

jrkerns, 2023-06-08:
Thanks for the find and the fix. It also appears that the crop method needs to be updated. E.g. doing `img.crop(pixels=0, ...)` will set the array shape to 0 which is unintended behavior. 

I will change this line to be `<= 0`: 

https://github.com/jrkerns/pylinac/blob/0b6d63f1cbb98743fedd3fbcde02ae33f33d3568/pylinac/core/image.py#L430

jrkerns, 2023-08-15:
Closing. Thanks for the contribution.

closed 2023-08-15 by jrkerns