jrkerns/pylinac #460: Problems install

ruipe90, 2023-07-20:
Hello,
I am trying to install pylinac but i am having problems and i don't know how to fix it. 
When I write pip install pylinac in anaconda prompt, it appears that: 

WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000002242C630FD0>, 'Connection to pypi.org timed out. (connect timeout=15)')': /simple/pylinac/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000002242B1733D0>, 'Connection to pypi.org timed out. (connect timeout=15)')': /simple/pylinac/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000002242C6521D0>, 'Connection to pypi.org timed out. (connect timeout=15)')': /simple/pylinac/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000002242C668A10>, 'Connection to pypi.org timed out. (connect timeout=15)')': /simple/pylinac/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000002242C63CDD0>, 'Connection to pypi.org timed out. (connect timeout=15)')': /simple/pylinac/
ERROR: Could not find a version that satisfies the requirement pylinac (from versions: none)
ERROR: No matching distribution found for pylinac

crcrewso, 2023-07-20:
Looks like pip couldn't find pypi.org, so it's not a pylinac issue. 
Try `pip install wheel`

ruipe90, 2023-07-25:
Thank you. Now I have fix it but when I have this lines:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Usuario\AppData\Roaming\Python\Python311\site-packages\pylinac\__init__.py", line 12, in <module>
    from pylinac.ct import CatPhan504, CatPhan600, CatPhan503, CatPhan604
  File "C:\Users\Usuario\AppData\Roaming\Python\Python311\site-packages\pylinac\ct.py", line 29, in <module>
    from .core import image
  File "C:\Users\Usuario\AppData\Roaming\Python\Python311\site-packages\pylinac\core\image.py", line 21, in <module>
    from .utilities import is_close, minmax_scale
  File "C:\Users\Usuario\AppData\Roaming\Python\Python311\site-packages\pylinac\core\utilities.py", line 2, in <module>
    from collections import Iterable
ImportError: cannot import name 'Iterable' from 'collections' (C:\ProgramData\anaconda3\Lib\collections\__init__.py)

I have a problem because cannot import name 'Iterable'. Had anyone had this error?

crcrewso, 2023-07-25:
I don't know if I've done any testing with Python 3.11. I'd recommend installing 3.10 and seeing what happens

LuisOlivaresJ, 2023-07-25:
Related to Cody Crewson's response, I recommend creating a new
environment to use python 3.10 and pylinac. To do that, use the terminal or
an Anaconda Prompt:

1.- To create an environment with a specific version of Python:

conda create -n myenv python=3.10

2.- Activate the new environment:

conda activate myenv

3.- Install pylinac:

pip install pylinac

For details, this is the environment user guide
<https://conda.io/projects/conda/en/latest/user-guide/tasks/manage-environments.html#>

jrkerns, 2023-07-26:
@ruipe90 What version are you trying to install? This issue was fixed in 2021: https://github.com/jrkerns/pylinac/commit/8de1b54a366c7ae3e19aa3af99cb1282860b3dfc#diff-a59a5d128ea7943015f7f559429a5ede8c9f35de4e8e1164b6bd7f6d91e124cc

jrkerns, 2023-08-15:
Closing due to being stale

closed 2023-08-15 by jrkerns