jrkerns/pylinac #465: Quartz Modified Phantom for Hypersight machines

JustinThymePhysics, 2023-09-19:
**Is your feature request related to a problem? Please describe.**
Hypersight Varian feature on Halcyon - has a new HU region for water.

**Describe the solution you'd like**
A new ROI "circle" of water material has been added to the Quartz model for machines with Hypersight option.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
[2023.08.28.zip](https://github.com/jrkerns/pylinac/files/12665521/2023.08.28.zip)

**Additional context**
Add any other context or screenshots about the feature request here.

JustinThymePhysics, 2023-09-19:
[SEPT2023Hypersight_HALCBCT_Paterson_py3__Body_CT.pdf](https://github.com/jrkerns/pylinac/files/12665532/SEPT2023Hypersight_HALCBCT_Paterson_py3__Body_CT.pdf)

jrkerns, 2023-10-22:
Thanks for submitting. I should be able to work with this, although if there's a manual that would be helpful as I can reference it in the documentation.

JustinThymePhysics, 2023-10-27:
[image]
[image]
[image]
[image]
[image]
[image]

Some relevant pages that discuss the HU topic and phantom revisions attached.
Thanks.
Not sure my email went through but here is my repeated message:
I was able to modify quart source:  I did -45 degrees and double the radius size worked fine and defined a new material called water which set to 0 HU.
But Page 7 says diameter of existing HU samples are 15 mm and the water is 35 mm.

jrkerns, 2023-11-09:
On its way and will be in the next release.

jrkerns, 2023-11-27:
https://pylinac.readthedocs.io/en/latest/quart.html#hypersight

closed 2023-11-27 by jrkerns