jrkerns/pylinac #478: Low priority - cleaning up Winston Lutz analysis and units?

ethanio12345, 2024-01-16:
Inconsistent use of where units are printed in the Winston Lutz results. Below represents some results for my exam prep in testing how sensitive various QA tests are with bad setups, so these aren't clinical results. :P 

Mean 2D CAX->BB distance: 0.70mm
Maximum Gantry RMS deviation (mm): 0.87mm
Maximum Couch RMS deviation (mm): 1.19

So, we have three different conventions where the units are (units in/not in description and units in/not in results). Is there a particular convention wanted to make it consistent? I.e. units only in results or description or units in both? I know this will be low priority, but would be a fairly quick/easy fix, in changing the relevant f-strings, and I'd be happy to make another commit.

jrkerns, 2024-01-16:
Hmm. Good question. I never really paid attention and clearly didn't think about it too much. It was probably based on what I had for lunch any given day. I don't much have a preference; however, in RadMachine we represent things as " label : value " so from that perspective the rubric " label (unit): value " would match that pretty well I think. 

[image]

Edit: GH hates brackets apparently

ethanio12345, 2024-01-16:
Hahaha, as I said, low priority issue, and not one which impacts actual results (which is the important part). I'll try and get a pull request ready on lunch break today following "label (unit): value"...if not, I'll probably forget. Thanks James.