jrkerns/pylinac #480: ACR MRI Slice Thickness Results

EricLessard, 2024-01-17:
**Describe the bug**
ACRMRILarge results for slice thickness reports dicom header slice thickness instead of measured slice thickness

in acr.py lines 1242-1255
```
def results(self, as_str: bool = True) -> str | tuple:
        """Return the results of the analysis as a string. Use with print()."""
        string = (
            f" - {self._model} Results - ",
            f"Geometric Distortions: {self.geometric_distortion.distances()}",
            f"Slice Thickness: {self.slice1.slice_thickness:2.2f}mm", 
            f"Slice 1 S/I Position shift: {self.slice1.slice_shift_mm:2.2f}mm",
            f"Slice 11 S/I Position shift: {self.slice11.slice_shift_mm:2.2f}mm",
            f"Uniformity PIU: {self.uniformity_module.percent_image_uniformity:2.2f}",
            f"Percent-signal ghosting: {self.uniformity_module.psg:2.2f}%",
            f'Uniformity Center ROI standard deviation:{self.uniformity_module.rois["Center"].std:2.2f}',
            f"Row-wise MTF 50% (lp/mm): {self.slice1.row_mtf.relative_resolution(50):2.2f}",
            f"Column-wise MTF 50% (lp/mm): {self.slice1.col_mtf.relative_resolution(50):2.2f}",
        )
```

The line
`f"Slice Thickness: {self.slice1.slice_thickness:2.2f}mm", `
Should be
`f"Slice Thickness: {self.slice1.measured_slice_thickness_mm:2.2f}mm",`

slice_thickness is set in the parent class CatPhanModule (ct.py) during initialization as
`self.slice_thickness = catphan.dicom_stack.metadata.SliceThickness`

As is, the printed results to console and pdf will always show 5mm. I tested this by modifying the dicom header to different slice thickness values.

jrkerns, 2024-01-17:
Yikes! Thanks for the heads up. I'll add it to the next release.

ethanio12345, 2024-01-17:
Also, just to follow on from my naming issue, if wanting to keep the naming convention the same as RadMachine (label (unit): value), probably could instead be;

``f"Slice Thickness (mm): {self.slice1.measured_slice_thickness_mm:2.2f}"``

Also could probably do the same thing for all the lines in the results.

jrkerns, 2024-02-22:
Fixed in https://github.com/jrkerns/pylinac/commit/1a15b61ea1d5393dee2e53f4534fd782a75ac590

closed 2024-02-22 by jrkerns