jrkerns/pylinac #481: Unable to locate Catphan 504

JianJuly, 2024-02-19:
Hello,

The CatPhan module (504) is unable to handle the following CBCT:
[20240218.zip](https://github.com/jrkerns/pylinac/files/14325566/20240218.zip)

Would you kindly assist me with this?

[image]

jrkerns, 2024-02-22:
What version are you using? On the master branch it says the scan extent isn't enough (i.e. not all modules were scanned). 

See here for scan instructions: https://pylinac.readthedocs.io/en/latest/cbct.html#image-acquisition

JianJuly, 2024-02-23:
> What version are you using? On the master branch it says the scan extent isn't enough (i.e. not all modules were scanned).
> 
> See here for scan instructions: https://pylinac.readthedocs.io/en/latest/cbct.html#image-acquisition

Thank you for. I've upgraded to the latest version 3.17.0, yet encountered the same error as you described. I will proceed to re-scan the module.

> ValueError: The physical scan extent does not match the module configuration. This means not all modules were included in the scan. Rescan the phantom to include allrelevant modules, or remove modules from the analysis.

closed 2024-02-23 by JianJuly