jrkerns/pylinac #484: Dynalog MU Information Not Extracted Correctly

1467822043, 2024-02-27:
**Describe the bug**
I am trying to use the pylinac library to convert Dynalog log files to DICOM files. However, I have noticed that the extracted MU values from the Dynalog log files always seem to be in the range of 0-25000, and I am unsure if this is indicative of missing MU information or if there's a specific method to retrieve the correct MU information.

[image]

jrkerns, 2024-02-27:
Per the Dynalog File format specification document (2013):

RS14678, 2024-02-28:
Thanks James，
Does the statement "all dose fractions range from 0 to 25000" indicate that the MU values retrieved from axis_data.mu.actual are inherently limited or devoid of meaning? Is there a specific calculation method to accurately derive MU information from Dynalog logs? I am eager to receive guidance on extracting the actual MU values. Looking forward to your reply.

jrkerns, 2024-02-28:
It means the MU are all relative. You'd have to know the MU from, say, the RT plan and then normalize the relative MU to the planned MU.

RS14678, 2024-02-29:
Thank you so much, according to the method you provided, I have got the data I want.

closed 2024-04-02 by jrkerns