jrkerns/pylinac #494: Multiple CT series not sorted properly into a Dicom stack

sarytky, 2024-05-17:
Thank you for the excellent library for physics in radiation therapy! We are currently automating our QA workflow in Oulu, Finland, and Pylinac has been integral in the process.

**Describe the bug**
In case there are images loaded from multiple series with the image.DicomImageStack class, the sorting does not work properly. Since the metadatas variable (see the screenshots below) changes size, self.image_path_keys updates incorrectly.

**To Reproduce**
We encoutered the behavior when exporting multiple series from Eclipse. The import export tool saves all images in the same folder, mixed.

**Expected behavior**
The function should return the sorted paths from the specific UID. This could be solved with some fixes in the code (see the second screenshot below)

**Screenshots**
Code from pylinac 3.22. The metadatas changes size with multiple series, while paths stays the same size.
[image]

Modifying also the paths variable accordingly, the sorting problem should be avoided.
[image]

jrkerns, 2024-05-17:
Ah, thanks for the heads up!

jrkerns, 2024-07-23:
Will be fixed in 3.26. Many thanks for the suggested code. It worked perfectly.

closed 2024-07-23 by jrkerns