jrkerns/pylinac #501: Missing data in XIM image when using Pelvis X-Large mode on Truebeam linac

bp-mp, 2024-06-27:
I'm using the pylinac.core.image.XIM module to read XIM files from a Truebeam linac. I was able to read most files, however some appear to have incorrect values in the final image. 

I was measuring narrow, in-air profiles captured when scanning a CTDI phantom. The XIM projections should look like a narrow beam of radiation, with a well defined peak when measuring a vertical profile. 

When measuring full-fan Head mode, the images looked correct (first image/profile)

When measuring half-fan X-Large Pelvis mode, the part of the image with the most intense radiation appears to be 0 (second image/profile). To me it looks like the data is missing? 

I have analysed the XIM files on the linac itself, and the images/profiles look correct on there. 

[image]

jrkerns, 2024-06-27:
Looks like possible datatype overflow error (e.g. the image might be 32 bit instead of 16). It's also possible that the specification has been updated since the original from Varian. If you're willing to submit the xim image I'll take a look. https://forms.gle/iyvf4wk7W2wr3oae6

jrkerns, 2024-07-23:
Got the image, thanks! Will take a look shortly.

jrkerns, 2024-07-23:
Yup, datatype overflow. The assumption in the original explanation was that there would only be uint8 or uint16 datatypes, but this file is trying to specify uint32. Easy fix. Will be in the next version. Let me know if you need a workaround sooner and I can get you a workaround.

bp-mp, 2024-07-31:
Great, thank you for taking the time to look at it!

 I should be ok until the next version, I can just analyse it manually on the linac.

closed 2024-07-31 by jrkerns