jrkerns/pylinac #503: CATPHANTOM ANALYSIS

Rahul-py-linac, 2024-07-12:
CAT PHANTOM CP604 MODLE 
The CAT Phantom 604 has contradicting in its model of different sections in the source code.
would kindly update with the new model which would be really use full for analyzing the CAT Phantom.
[image]
[image]

crcrewso, 2024-07-17:
By contradiction would you like to see that the names of the modules from the manuals instead of the modules as related to the 504 phantom?

Rahul-py-linac, 2024-07-19:
Yes, the CATPHAN604 sections shown above are different from the modules mentioned in the source code.

crcrewso, 2024-07-22:
> Yes, the CATPHAN604 sections shown above are different from the modules mentioned in the source code.

I agree, the proper names should be used, and alias' added for historical compatibity (technical and human)

jrkerns, 2024-07-23:
Yes, this is a very old convention that was meant so code could be adjusted more easily by only adjusting the initial class constructor. I.e. you could always get to the uniformity section by doing `.ctp486`, etc. You'd be surprised how many users conflate the 604 with the 504, so being able to just switch the constructor was actually advantageous for that issue. The human-readable naming is easy to change. Yes, creating aliases is definitely the way to go for compatibility. Only downside is having, say, 8 modules in `results_data` instead of the expected 4 (all RadMachine customers use the output from `results_data` for reference and tolerance checking). I will put a ticket in our system to see if the PM concurs with that flow. Thanks!