jrkerns/pylinac #504: Date and time of acquisition of Image

Rahul-py-linac, 2024-07-12:
It would be great if the pdf containing the results would also have the time and date of image acquisition. 
Hope that make sense
Thankyou 
Regards

crcrewso, 2024-07-22:
I can understand where this is coming from. For my uses it doesn't matter, for multi-image analysis like W-L do you have an argument for the first or last timestamp to be used?

jrkerns, 2024-07-23:
The datetime could be passed via the `notes` argument as a workaround. 

`<wl>.publish_pdf(..., notes="...")`

Obviously, that's not ideal. I would certainly hope that the images are taken at roughly the same time which would mean the first or last datetime are similar. I would probably word it something like "Last image datetime of xxxxxxx" adjacent to the "Generated with Pylinac ..." statement.