jrkerns/pylinac #506: Winston Lutz keyed_image_details keys have decimal places

adamblais, 2024-07-19:
**Describe the bug**
The dictionary obtained using wl.results_data() has keys that have decimal places which make the keys useless. For example, the documentation says to access the dictionary using a key such as 'G0B90P0'. However, the keys in my dictionary appear as: 'G0.006634599471B90.0003014433343P0.01171875'
or
'G359.980094726117B44.9996175082791P359.986328125'

I assume it is getting the gantry and other values from the dicom images and these will have different values whenever the gantry, collimator or couch are moved. Is it possible to add rounding to these dictionary keys? Currently I cannot use a key such as 'G0B90P0' so it is not useful.

I am using dicom images acquired on a TrueBeam.

(as an aside... why does pylinac use B and P to refer to collimator and couch/table? Seems confusing...)

jrkerns, 2024-07-19:
Yeah I can round that to the nearest whole number. 

B (beam collimation) and P (patient position rotation) were conventions from very early on that I adopted to avoid overlap with a commercial competitor at the time to avoid litigation. Bad choices are hard to change after they've been in use for a while.

jrkerns, 2024-07-23:
I forgot, you can already do this with the `axes_precision` keyword in the constructor. It defaults to None (no precision definition), but if you set it to 0 it will round to the integer. Here's a standalone example.

```
from pylinac.core.image_generator import AS1200Image, generate_winstonlutz, FilteredFieldLayer
from pylinac import WinstonLutz

files = generate_winstonlutz(
    simulator=AS1200Image(1000),
    dir_out='wlprecision',
    field_layer=FilteredFieldLayer
)

wl = WinstonLutz(directory='wlprecision', axis_mapping={
    files[0]: (0.0012, 0.1234, 0.1111),
    files[1]: (45.013, 0.1234, 0.1111),
    files[2]: (89.09, 0.1234, 0.1111),
    files[3]: (359.995, 0.1234, 0.1111),
}, axes_precision=0)  # note this line -----------------------
wl.analyze()
d = wl.results_data().keyed_image_details
# 0.0012 -> G0B0P0, ...
```

adamblais, 2024-07-23:
Using axes_precision in the constructor works great, thank you!

closed 2024-07-23 by jrkerns