jrkerns/pylinac #508: CatPhan 604 Analysis with Direct Density CT images

adamblais, 2024-07-26:
I'm trying to analyze a CatPhan 604 scan acquired on a Siemens CT scanner using their direct density (DD) algorithm. The standard (non-DD) scan works, but the direct density fails to find the HU linearity module. I believe the issue is the significantly different calibration curves.

I tried overriding the HU's using the custom HU's feature based on the HU values reported in the different landmarks in ARIA:

```
mycbct.analyze(expected_hu_values={
    "Teflon": 468,
    "Delrin": 162,
    "20% Bone": 100,
    "Acrylic": 80,
    "Air": -999,
    "Poly": -25,
    "LDPE": -80,
    "50% Bone": 311,    
    "PMP": -170
    })
```

But it still fails to find HU linearity module. Anything else I can try? Is this possible? Happy to provide a sample dicom image if it helps.

jrkerns, 2024-07-26:
https://forms.gle/jf7DjGRQb98nw6W66

keithoffer, 2024-09-09:
We recently installed a Siemens scanner and I noticed the same thing. I work at a different site from where it was installed, so I'm not fully across it, but I noticed the HU values in the high density inserts were quite a bit lower than what I've seen in the past for other scanners. In the end I just adjusted the hu_origin_slice_variance from the default 400 to 200 for just that scanner to let them analyse but not touch the analysis for any of our other scanners.

Something like:
```
cbct = CatPhan.from_zip(catphan_upload)
cbct.hu_origin_slice_variance = 200

# Continue with the rest of your code as normal
```

I didn't want to raise it as an issue yet as I'm still confused as to why the reported HU are so much lower in the first place.

adamblais, 2024-09-10:
Thanks! I did get it working with @keithoffer's suggestion of adding
`cbct.hu_origin_slice_variance = 200`

I was just looking at the white paper as well. It looks like the Direct Density images do not use HU values, but rather relative electron density values, so they are meant to be different.

jrkerns, 2024-09-10:
Yes, that's the workaround for the "can't find HU module" problem. Another clinic is having issues with wrong slices being analyzed, partially due to this variance, which causes the "origin slice" to be off by several slices vs failing outright. I will add a small note about this algorithm to the docs. Thanks for bring it to our attention.

NickyPooHFR, 2024-09-24:
Hello,
We have a CatPhan 600 phantom, and although changing the cbct.hu_origin_slice_variance value to something like 200 or 300, 
There's still an issue. The ValueError "Unable to locate the Geometric nodes" is raised, and the solution is less obvious. Any idea on how to proceed ?
Happy to provide DICOM files if necessary !

jrkerns, 2024-09-24:
Yes please. You can submit here: https://forms.gle/fWYCXJDgkjxdR8vL6

NickyPooHFR, 2024-09-25:
> Yes please. You can submit here: https://forms.gle/fWYCXJDgkjxdR8vL6

Done ! Thanks for your help !

jrkerns, 2024-10-04:
Unfortunately, any correction I apply cause other tests to fail. I will try a different node-finding strategy.

jrkerns, 2024-10-04:
Fix is at code review. Should be included in the next release. Will still need to set the HU origin variance but will work after that.