jrkerns/pylinac #513: linux in untubu 20.04 ACR Phantoms analyse error

djp82645, 2024-09-03:
hi,i have some problem use by linux.

it's work well in windows.but in linux have some error and diff in windows so well.
error like this:
1 Could not determine the roll of the phantom. Ensure the 20mm top-left circle is visible on Slice 1
2 The physical scan extent does not cover the extent of module configuration. This means not all modules were included in the scan. Rescan the phantom to include all relevant modules, or change the offset values.

file and analyse results for example:
windows results:
- ACR MRI Large Results -
Geometric Distortions: {'horizontal': '19.53mm', 'vertical': '18.56mm', 'negative diagonal': '11.05mm',
'positive diagonal': '12.43mm'}
Slice Thickness: 5.19mm
Slice 1 S/I Position shift: -0.49mm
Slice 11 S/I Position shift: 0.00mm
Uniformity PIU: 86.16
Percent-signal ghosting: 0.02%
Uniformity Center ROI standard deviation: 338.24
Row-wise MTF 50% (lp/mm): 1.03
Column-wise MTF 50% (lp/mm): 0.97

linux results:
- ACR MRI Large Results -
Geometric Distortions: {'horizontal': '189.46mm', 'vertical': '167.00mm', 'negative diagonal': '189.21mm',
'positive diagonal': '186.38mm'}
Slice Thickness: 2.12mm
Slice 1 S/I Position shift: 0.00mm
Slice 11 S/I Position shift: 0.00mm
Uniformity PIU: 85.54
Percent-signal ghosting: 0.01%
Uniformity Center ROI standard deviation: 353.50
Row-wise MTF 50% (lp/mm): 0.69
Column-wise MTF 50% (lp/mm): 0.85

analyse file:
[QA-2024028.zip](https://github.com/user-attachments/files/16842266/QA-2024028.zip)
Looking forward to your reply.
Thank you！

jrkerns, 2024-09-04:
Haven't seen that one before. Can you share your numpy and scipy versions from each environment? That might also help narrow it down. Thanks.

djp82645, 2024-09-25:
> Haven't seen that one before. Can you share your numpy and scipy versions from each environment? That might also help narrow it down. Thanks.
linux
scipy==1.11.3 
numpy==1.26.1
pylinac==3.20.0
windows
scipy==1.11.3 
numpy==1.26.1
pylinac==3.20.0
the environment of numpy and scipy versions  is same.

djp82645, 2024-09-26:
> > Haven't seen that one before. Can you share your numpy and scipy versions from each environment? That might also help narrow it down. Thanks.
> > linux
> > scipy==1.11.3
> > numpy==1.26.1
> > pylinac==3.20.0
> > windows
> > scipy==1.11.3
> > numpy==1.26.1
> > pylinac==3.20.0
> > the environment of numpy and scipy versions  is same.

or can you give me that pylinac docoumt's file.I will test it in windows and linux. 
[image]

jrkerns, 2024-09-26:
That's just a made-up file name for demonstration purposes.

djp82645, 2024-09-27:
> That's just a made-up file name for demonstration purposes.

well,but why have error analyse in liunx. that I test in two manchine in liunx and environment is same use the same file.but one analyse suceesful.but other is error,error is:Could not determine the roll of the phantom. Ensure the 20mm top-left circle is visible on Slice 1.