jrkerns/pylinac #515: Wrong variable name in TG51ElectronModern

mchamberland, 2024-09-05:
**Describe the bug**
TG51ElectronModern fails to return an adjusted output because the wrong attribute name is used on [line 1319](https://github.com/jrkerns/pylinac/blob/37319ba843441a7f0f0a1acac356f4650c8976e9/pylinac/calibration/tg51.py#L1319). This should be `m_corrected_adjustment`.

**To Reproduce**
Create a TG51ElectronModern instance with all needed parameters, including `m_reference_adjusted`. Then, try to access `dose_mu_dmax_adjusted`.

**Expected behavior**
Should return a calculated adjusted dose per MU at dmax.

**Additional context**
Python output: 

> Invalid Test Procedure: tg_51_electrons_output_adjustment_wgtg51_e", line 28, in tg_51_electrons_output_adjustment_wgtg51_e
>   File "/usr/src/qatrackplus/.env/lib/python3.11/site-packages/pylinac/calibration/tg51.py", line 1328, in dose_mu_dmax_adjusted
>     return self.dose_mu_dref_adjusted / (self.clinical_pdd / 100)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/src/qatrackplus/.env/lib/python3.11/site-packages/pylinac/calibration/tg51.py", line 1319, in dose_mu_dref_adjusted
>     * self.m_corrected_adjusted
>       ^^^^^^^^^^^^^^^^^^^^^^^^^
> AttributeError: 'TG51ElectronModern' object has no attribute 'm_corrected_adjusted'

jrkerns, 2024-09-05:
Thanks for reporting. Looks like I made a typo. 

Until v3.27 that comes out ~Sept 19th you should be able to work around this by overloading the property like so:

```
from pylinac.calibration.tg51 import TG51ElectronModern

class FixedTG51ElectronModern(TG51ElectronModern):
    
    # dose dmax depends on the dose at dref which is where the problem is
    @property
    def dose_mu_dref_adjusted(self) -> float:
        """cGy/MU at the depth of Dref."""
        return (
            self.tissue_correction
            * self.m_corrected_adjustment  # here's the fix; was m_corrected_adjusted; typo
            * self.kq
            * self.n_dw
            / self.mu
        )
    
e51 = FixedTG51ElectronModern(
    ...
)

dose_adjusted = e51.dose_mu_dmax_adjusted
```

mchamberland, 2024-09-05:
Great, this works. Thank you!

closed 2024-09-11 by jrkerns