jrkerns/pylinac #520: Calibration protokol TRS398  k_t_p  calculation

bozhikovstanislav, 2024-10-13:
**Describe the bug**
A clear and concise description of what the bug is.
the formula for k_t_p --- for TG51
   return ((273.2 + temp) / 295.2) * (101.33 / press)

the formula for k_t_p - for TRS38
   return ((273.2 + temp) / 293.2) * (101.33 / press)

it is around 1% difference - most SSDL in EU do calibration on 20 C 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
Add any other context about the problem here.
[image]

jrkerns, 2024-10-14:
I definitely missed the difference in reference temperature! Really sorry about that. I'll have a fix and warning in the docs.

jrkerns, 2024-10-15:
Will be fixed in v3.29!

jrkerns, 2024-10-28:
https://pylinac.readthedocs.io/en/latest/changelog.html#trs-398

closed 2024-10-28 by jrkerns