jrkerns/pylinac #534: CT Metadata access

crcrewso, 2024-12-18:
**Is your feature request related to a problem? Please describe.**
Currently, analysis of CT/CBCT is impacted by uncertainty in the energy and current used for the data collection. 

**Describe the solution you'd like**
I would like to be able to retrieve the DICOM header of the central slice for any CT/CBCT dataset as a property. 

**Additional context**
I am unsure if it would be best to:
- add the full DICOM metadata to the main property, this could be problematic if it is called before analyze()
- add it as a value to the results_data object, 
- create a list of the properties of interest and add that subset only

jrkerns, 2024-12-30:
The DICOM tags are available in the `dicom_stack` object. If you know the slice ahead of time you can access individual slices by doing:

```
ct = CatPhan504.from_demo_images()
print(ct.dicom_stack[11].metadata.Exposure)
```

You could do the same after analysis by doing:

```
ct = CatPhan504.from_demo_images()
ct.analyze()
data = ct.results_data()
print(data.origin_slice)
print(ct.dicom_stack[data.origin_slice].metadata.Exposure)
```

crcrewso, 2024-12-30:
I can see how that would work and that I was missing a specific detail in my original post. 
There's no corresponding key for the results_data(as_dict=true) 

Do you have an idea for how we could add these parameters to the results_data dict object easily and consistently?