jrkerns/pylinac #541: Unit tests fail with "KeyError: 'area_bbox'"

mivermeu, 2025-02-05:
**Describe the bug**
Multiple existing unit tests fail with the message "KeyError: 'area_bbox'"

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the most up-to-date version of the project (tested on commit 75e1c2a8869d3a34a2ba2e924d807d1a11270911)
2. Install the requirements with `uv sync --frozen`
3. Run the test suite with `uv run pytest`

**Expected behavior**
I expected all unit tests to pass with a clean install of the project.

**Additional context**
Sample stack trace:
```
_______________________________________________________________________________ ERROR collecting tests_basic/test_winstonlutz_mtmf.py _______________________________________________________________________________
tests_basic/test_winstonlutz_mtmf.py:10: in <module>
    from pylinac import WinstonLutzMultiTargetMultiField
pylinac/__init__.py:12: in <module>
    from .acr import ACRCT, ACRMRILarge
pylinac/acr.py:20: in <module>
    from .core.mtf import MTF
pylinac/core/mtf.py:15: in <module>
    from .roi import HighContrastDiskROI
pylinac/core/roi.py:9: in <module>
    from skimage.measure._regionprops import _RegionProperties
.venv/lib/python3.13/site-packages/skimage/measure/_regionprops.py:1426: in <module>
    _install_properties_docs()
.venv/lib/python3.13/site-packages/skimage/measure/_regionprops.py:1421: in _install_properties_docs
    getattr(RegionProperties, p).__doc__ = prop_doc[p]
E   KeyError: 'area_bbox'
```

jrkerns, 2025-02-05:
Looks like I need to bump pyproject to scikit-image 0.19+. v0.18 (2021 and prior) had different default names for region props: 

[image]

https://scikit-image.org/docs/stable/release_notes/release_0.19.html

I'm surprised uv would default to such an old scikit-image version. You should be able to bump scikit-image to 0.19+ to get around this.

mivermeu, 2025-02-06:
It seems that a combination of scikit-image==0.24.0 and python3.13 was the culprit. I tried installing just this version of scikit-image, after which attempting to load regionprops failed with the same KeyError.

Steps to reproduce:
```bash
uv venv -p 3.13
source .venv/bin/activate
uv pip install scikit-image==0.24.0
python -c "from skimage.measure import regionprops"
```

Either forcing python3.12 or scikit-image>=0.25.0 (the latter requires python>=3.10) solves the issue. However, new errors appear when running `uv run pytest`. Sample stack trace:
```
tests_basic/test_plan_generator.py:27: in <module>
    RT_PLAN_FILE = get_file_from_cloud_test_repo(["plan_generator", "Murray-plan.dcm"])
tests_basic/utils.py:124: in get_file_from_cloud_test_repo
    with access_gcp() as client:
/opt/homebrew/Cellar/python@3.12/3.12.5/Frameworks/Python.framework/Versions/3.12/lib/python3.12/contextlib.py:137: in __enter__
    return next(self.gen)
tests_basic/utils.py:45: in access_gcp
    client = storage.Client.from_service_account_json(str(credentials_file))
.venv/lib/python3.12/site-packages/google/cloud/client/__init__.py:107: in from_service_account_json
    credentials_info = json.load(json_fi)
/opt/homebrew/Cellar/python@3.12/3.12.5/Frameworks/Python.framework/Versions/3.12/lib/python3.12/json/__init__.py:293: in load
    return loads(fp.read(),
/opt/homebrew/Cellar/python@3.12/3.12.5/Frameworks/Python.framework/Versions/3.12/lib/python3.12/json/__init__.py:346: in loads
    return _default_decoder.decode(s)
/opt/homebrew/Cellar/python@3.12/3.12.5/Frameworks/Python.framework/Versions/3.12/lib/python3.12/json/decoder.py:337: in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
/opt/homebrew/Cellar/python@3.12/3.12.5/Frameworks/Python.framework/Versions/3.12/lib/python3.12/json/decoder.py:355: in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
E   json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

After failing to run the tests, an empty file `GCP_creds.json` can be found in the project root.

jrkerns, 2025-02-06:
More info here: https://forum.pylinac.com/t/running-pylinac-unit-tests-fails/492

closed 2025-02-13 by jrkerns