jrkerns/pylinac #546: XIM Reader speedup

purepani, 2025-02-17:
**Is your feature request related to a problem? Please describe.**
Currently the speed of the XIM parser is quite slow due to being mostly written in pure python, and also partially being written with some missing optimizations(I.e. converting numbers to strings). 

**Describe the solution you'd like**
I've began writing an XIM parser in rust: https://github.com/purepani/xim_reader. This is able to be imported as a python package through `pyo3`. It isn't fully finished(it only supports `i16` at the moment, but I've gotten it to work well enough to run some tests, and it reduces the read time from ~1.1s to 7.5 ms: 
[image]

If this is something you'd like to add, I think it'd be best to vendor it, especially since you recently switched to `uv` with some nice workspace features. I'd be willing to finish it up and vendor it if it's something you'd like to add. 

**Describe alternatives you've considered**
I also tried speeding up the parsing by just improving the python and trying to do more array operations. This, however, was much harder. I got some speed up with using packbits, but to improve the main decompression algorithm, I'd have to be a lot more clever with array operations, whereas the rust implementation is basically a mirror of the python version.

purepani, 2025-02-23:
The parser should support multiple datatypes now, and now actually gives the histogram and properties. I unfortunately only have my one image to test with, so I'm not 100% sure it works with different types, but it should be pretty easy to modify if not.
I'll try and make a PR at some point

jrkerns, 2025-02-24:
This is pretty cool! I knew Rust would be a great tool for image and file parsing. I can imagine Trajectory log reading and gamma calculations also being some good targets. Unfortunately, I don't know much about Rust yet. Also, the Github CI/CD isn't set up because this is just a mirror for our company repo in bitbucket, although we've been in talks to get migrated over. 

At this point I don't know that I can maintain a Rust implementation for our RadMachine customers if there was, say, a bug. The overall speed improvements are flabbergasting as companies like Astral have shown. 

I honestly think such a tool is useful in its own right and rather than wait on me (an old, slow dinosaur), I would actually encourage you to package it up as its own PYPI package. Then other libraries, including pylinac, could call or install it. E.g. we could add an optional dependencies section with this package in it and if installed will use this much faster implementation. 

This gives me some motivation to look at Rust and also see about getting the Github CI/CD on parity with our bitbucket CI/CD flows.

purepani, 2025-02-24:
This is about what I expected as a response haha.
I will probably go ahead and package it up then when I'm able(and if I can get some images for testing).

I will note one thing about the gamma calculations though since you mentioned them(which prompted me to take a look); while certainly writing it in rust would create a big speedup, it could probably be reimplemented using `scipy.generic_filter`(i'm not sure this would be much faster without a C-extension) or (assuming this gets merged) this updated, much faster version of the function: https://github.com/scipy/scipy/pull/22575.

jrkerns, 2025-02-24:
Send me an email and I'll get you some .xim images from my old clinic.

purepani, 2025-02-24:
pani@wustl.edu will work

purepani, 2025-02-27:
@jrkerns I uploaded a version of the package under `xim-reader-rs` if you want to try it out experimentally. I also updated the readme with some instructions.

jrkerns, 2025-02-27:
Awesome! I will also add the project to https://github.com/jrkerns/awesome-medphys

purepani, 2025-02-28:
If you ever want a PR with this, I can add it as an optional dependency pretty easily since I am already forking pylinac to use it.

purepani, 2025-03-11:
Closing this issue since it is sped up

closed 2025-03-11 by purepani