jrkerns/pylinac #561: Google Colab not able to run pylinac

joecastelo, 2025-03-24:
**Describe the bug**
When installing with pip I'm having the following error:
----> 1 import pylinac

9 frames
/usr/local/lib/python3.11/dist-packages/numpy/__init__.py in __getattr__(attr)
    365                 raise AssertionError()
    366         except AssertionError:
--> 367             msg = ("The current Numpy installation ({!r}) fails to "
    368                    "pass simple sanity checks. This can be caused for example "
    369                    "by incorrect BLAS library being linked in, or by mixing "

ModuleNotFoundError: No module named 'numpy.char'

ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new notebook on Google Colab
2. !pip install pylinac
3. import pylinac
4. See error

**Expected behavior**
Standard workflow, tested with version 3.31.0 too. It was working not to long ago...

jrkerns, 2025-03-24:
I got the same. When I restarted the session and run all everything worked:

[image]

joecastelo, 2025-03-24:
Cool, worked here too! For everyone with the same error. Ctrl+M. and then Run again. 
Restart and Run All command did not work for me.

joecastelo, 2025-03-24:
Should we close it?

closed 2025-03-24 by jrkerns