jrkerns/pylinac #563: Remove warning for tg51 addendum

purepani, 2025-04-09:
https://github.com/jrkerns/pylinac/blob/3165ae3b12672a8bc4595a500dee9cafa089a407/pylinac/calibration/tg51.py#L1229-L1233

This warning can (probably?) be removed since the new addendum came out last year. It should probably be confirmed that all the data is the same as in WGTG51-e though.

mchamberland, 2025-04-10:
I have personally verified all the data, for what it's worth.

purepani, 2025-04-17:
I was thinking of stuff more like what's actually included in the report vs the data available back then. I think WGTG51-e includes a more limited set of chamber calibration factors than what is included in pylinac at the moment.
Whether it actually matters to reflect this is up for debate, of course.